smagicld/xiaoxiangBMS #107: Limiting the max discharge amp

CSYGD, 2020-10-07:
Hi
To keep some autonomy for my ebike I would love to limit the maximum delivered power, the maximum output amperage in other ways. This can be particularly interesting for my smallest batteries.
Do you think this can be possible?
Best regards 
Yoann

TheDodgersOnline, 2020-10-07:
Hi

This is already available through the app (IOS version) in the BMS settings.
There is another setting for the % of max current you can use.

Regards

[image]

RaWolt, 2020-11-09:
One additional question: Does anyone test this?  Does the board cuts the charge or discharge rates to the limit I set it here 
or only stops the charging or discharging at all  (to 0 A )  until I reset this "error" manually?

Kind Regards
Ralf

CSYGD, 2020-11-23:
Hi!

Thanks for your fast answer and sorry for my very long one... I just found the time to test what you said right now. 

Yes as I already noticed it is possible to set as a protection a current discharge. After a period of time the battery will just cut. It is not what I was wondering: My aim is to limit the max current to a certain point, not to cut everything. Stop the bike, restart and initialize the controller and motor then start again to ride my bike...

Also you spoke about a percentage of max current. Sorry but I can not find what you are writing about. Thanks in advance for some precision. 

There is a special tab current calibration but the online manual is empty. I will be pleased to understand more about this. 

Do you thing limiting the max currently is easy to do on the BMS side or it is more something I need to check on the motor side? The problem is that I am using several batteries and the motor has no clue about the one plug to it. 

Have a beautiful week !

closed 2020-12-20 by smagicld