smagicld/xiaoxiangBMS #108: App does not connect to device

bwcoetzer, 2020-10-11:
I have ANT-BLE16S device and uses iPhone 8 with software v13.8.
The Xiaoxiang BMS Monitor app v1.2.19 does not connect to the BMS and indicates only "waiting for data".
I bought the in-app offering to control the BMS

I also have installed ANTBMS  from Apple App store (cannot find a version number).  This connects and the app controls the BMS.

Screenshots of the two apps attached
[image]
[image]

smagicld, 2020-10-11:
Hi

The app is made only for XiaoxiangBMS. 
Where did you bought the ant bms? 
I could try to add support for it.

bwcoetzer, 2020-10-13:
Apologies, I was under the impression that Xiaoxiang and Ant are the same thing with chinese and english names.  The ANTBMS app (App store) references site http://mayibms.com/index.html. It will be very good to have an app of which I can understand all the functions. Early assessment is that the BMS works well

bwcoetzer, 2020-10-14:
Some resources here incl protocol manual - https://drive.google.com/file/d/1E0HEes6kgBze5DI5MRO2mkE8BmNdzXas/view

h2onitrox, 2020-11-07:
I’m experiencing exactly the same problem as becoetzer. I Downloaded same apps on my iPhone and consequently purchased the Apple in store xiaoxiang BMS Monitor which does not connect to the BMS.

See attached photo of BMS model for your reference and please advice 

[image]

smagicld, 2020-11-07:
This board does not look like XiaoxiangBMS BMS, more like ant. 

Ant BMS Support comes later.

h2onitrox, 2020-11-07:
Thanks for the feedback appreciate. So how do I get a refund on the product not applicable to my BMS?

smagicld, 2020-11-07:
Ask Apple.

smagicld, 2020-12-20:
Just FYI: Implemented Ant BMS support. if you drop me your email I'll add you into the test group.

closed 2020-12-20 by smagicld

bwcoetzer, 2020-12-21:
Excellent!
My email bwcoetzer@sumanco.com<mailto:bwcoetzer@sumanco.com>

Ben Coetzer

Nivya958, 2022-06-09:
Hi, I am trying to connect QUCC BMS to xiaoxiang app. But I lost the bluetooth module that came with the BMS. I got a HC-05 module and I'm trying to use that to connect the BMS to the app. The app does not detect the BMS in the first place. Is it because, I'll have to configure the bluetooth module?

Sharkynos, 2026-10-18:
Hello, could I please be added to the test group as well? Thank you. Email: sharkynos@gmail.com