smagicld/xiaoxiangBMS #111: Restore purchase

PappaBen82, 2020-10-12:
I have a serious issue using this app without internet connection . I already paid for a free version . This is a serious issue . I turn battery off via App but what if I don’t have internet connection ? Is impossible to change parameters or the battery on without internet connection . This is not ok , please fix this issue .

smagicld, 2020-10-12:
Hi

Please give more details:
- iOS Version
- App Version and Build
- how to reproduce the problem

PappaBen82, 2020-10-12:
iOS 14.01/ bms App Version 1.2.19 Build 5328.  So the App is great and works fine except every time when I connect to bms and I want to make a change or turn on output -input mosfet , the app requires to restore the app . It works good to hit ”restore purchase ”button  but not if internet conection is missing . 

Skickat från min iPhone

smagicld, 2020-10-12:
Can you try to delete the app completely (from the iOS device) and install it once again and then restore purchases?

I have tried on five iOS devices to reproduce your problem - no issues.

PappaBen82, 2020-10-12:
I uninstalled several times without success but somehow after I turned on and of all connections an a phone restart seems to work now . Thank you , I love this app everything is clean and intuitive. Thanx .

Skickat från min iPhone

smagicld, 2020-10-15:
Can you please update your AppStore rating?

Thanks
Lukas