smagicld/xiaoxiangBMS #114: Unable to correctly enter capacity config

powneill, 2020-10-13:
For all items under Capacity Configuration except the top two, it only accepts a maximum of 4 digits.  Are the top two lines per-battery-pack and the rest of the lines per-cell?  If so, it should be clearly labeled as such.  If all lines in this section are per-battery-pack, then a bug exists where more than 4 digits are not accepted in most fields.

smagicld, 2020-10-13:
I see. in German language it is labeled as **Pack** Designed capacity and **Cell** full charge voltage and and...

I'll fix the english labels.

powneill, 2020-10-13:
Perfect, thank you for the quick response!

smagicld, 2020-10-14:
[image]

cleaned up the naming. will come with next release.

closed 2020-10-14 by smagicld