smagicld/xiaoxiangBMS #115: Charge- and discharge readings 0.01A

WiardBeek, 2020-10-17:
On my iPhone I see that values below 0.2A are not shown. The display then stays at 0.00A. Please show values as low as 0.01A. I charge 32 (4S8P) Li-ion cells by a 50Watt Solar Panel for charging my iPhone etc. daily. As soon as the charging is below 10mA I have to disconnect the Solar panel because during darkness the battery discharges by 100mA to the Solar Panel. So at 10mA I must get an alert for disconnecting the battery.

NeariX67, 2020-10-25:
The app can only display the data it gets. The BMS usually returns a rather inaccurate current measurement that may vary between 100-500mA steps. So it's rather an issue with the BMS software and its protocol.

WiardBeek, 2020-10-25:
Thanks for your reply, I will wait for a new BMS software / hardware version.

sarah1331sarah, 2022-09-04:
My tests also show upto and sometimes including 5w discharge or charge shows as 0 

kills me but just get along with it 

would love a hardware update