smagicld/xiaoxiangBMS #116: Different voltage profiles for charge and discharge

ulihuber, 2020-10-21:
The voltage values attached to percentage of charge do either work while charging -or- while discharging.
If there were two sets of values charge status would be displayed more accurately.

DLAWRY, 2021-07-02:
Hello,I updated cell voltages based on data sheets believing I would observe more accurate battery SOC %.  I used discharge data. I do not however see the corresponding SOC % results in the BMS monitor. Example: 80 % cell voltage shows 53% SOC. How does the app calculate SOC?
Thanks!
Dlawry

smagicld, 2021-07-02:
The app does not calculate SOC. It displays what the bms is reporting. If you have troubles setting correct parameters ask your bms or battery cell reseller to give you the correct values (or take the values from cell data sheets).

DLAWRY, 2021-07-03:
Hello,
Thank you, I now understand the BMS reports the data only and the app does no calculations.

I am using the cell data sheets from the manufacturer, and have now lowered the default “Cell full Voltage” parameter to be closer to the actual battery data for full charge (Fortune cells).  This resulted in a more accurate SOC for my configuration. Good.

I had previously only updated the “Cell [80,60,40,20] Cap. Voltage” capacity configuration parameters.

Regards,

Dlawry