smagicld/xiaoxiangBMS #117: Make the app opensource

NeariX67, 2020-10-30:
I'd really like to contribute to this project, for example helping with some logging issues that i found and some others bugs in v2.0.
To avoid that users build the app for themselves, i would remove relevant storyboards (if used, i guess smagicld used SwiftUI though), remove xcode workspace and xcode project, maybe remove other classes that reveal specific functionality.
This would allow us to add features like GPS speeds, efficiency, remaining range for EVs for example.

smagicld, 2020-11-02:
Hi. If you want open source - then write your own app and publish it.

The Idea of current App was not to have it Biker / Roller App. Just BMS configuration.

In Version 2 of the app (rewritten in SwiftUI) I have added support for different BMS Screens.

If you are still interested, I can build an app just with the demo adapter and BMS generic interface so you can create your own screen based on that. I could then add it into the app.

NeariX67, 2020-11-02:
I‘m already more than half way done with my own app. But i just wanted to avoid beeing a competitor and rather help advancing an existing app with existing users. The main problem i‘m facing is the Bluetooth connection itself, i can connect, discover and write, but when i read, the result is filled with 0x00‘s. I wrote a scetch for an ESP8266 that will redirect serial messages and UDP packets bidirectional, so i could test the app. If you‘ve read my email, any help regarding the bluetooth problem would be appropriated.

NeariX67, 2020-11-03:
Recoded the whole bluetooth class and it is working now. I guess it was a timing issue

closed 2020-12-20 by smagicld