smagicld/xiaoxiangBMS #118: Cannot turn BMS back on after turning off

wbraray, 2020-11-02:
Turned battery off via app now battery will not turn back on

smagicld, 2020-11-02:
any details?!?

wbraray, 2020-11-02:
[image]
[image]

wbraray, 2020-11-02:
No display on battery lcd

smagicld, 2020-11-02:
what is the alarm?

Looks like your output has an shortcut.has nothing to do with the app.

closed 2020-11-02 by smagicld

wbraray, 2020-11-02:
Hi 
No load connected, short alarm appeared on initial connection was possibly inverter trying to power up, all systems worked, turned off using app, but now will not turn back on. Have disconnected load and rebooted app, app-still appears to communicate with battery as it displays voltages etc. when u use the turn on button message asks if I want to turn on etc, but app still displays battery discharge turned off.
If not app is it the BMS ?
Thanks for the reply!