smagicld/xiaoxiangBMS #120: Changed battery capacity after changing the minimum charging temperature

AkiMax, 2020-11-18:
I have charged this great App because I want to adjust the minimum charging temperature of my batteries. It is a workaround, because they always show about 10° lower than the ambient temperature, over a long period of time.
After I changed the minimum charging temperature by 1° for a careful test and loaded the data back into the BMS, the battery capacity display has changed from 75% to 60%. 
I find this astonishing, as I did not change any other value. A comparison of all settings with the identical battery connected in parallel shows no differences either, only this strongly changed percentage value. 
I have two Liontron 12,8V / 200Ah LiFePO4 batteries, the connection with the app worked without any problems. 
Do you have an explanation for this?
Many thanks in advance, Axel

NeariX67, 2020-11-29:
I noticed this behaviour too. But my pack just went from 100% to 99-96% or something like that. This has nothing to do with the app, the cause of this issue is the BMS recalculating the state of charge based on voltage or whatever and it's therefore rather inaccurate

madysouma, 2020-11-29:
Even me too I have  the same problem when I just charge the battery to 100% and if I tried to enable balancing or changing something It's just re-calculate and dropped volt or ah 
Something is not okay with this Bms or need update ￼￼

jamu85, 2020-12-13:
Same for my 4s 100ah. When changing the values in the app, the capacity level or voltage levels change. Not sure if that’s a bug in the app or the bms firmware

smagicld, 2020-12-13:
The App itself has no such logic. It shows 1:1 values provided by the BMS.