smagicld/xiaoxiangBMS #121: 18s Unable to set > 17 cells

magharees, 2020-11-20:
The overall voltage is different than my multimeter + for some reason it won’t allow me to change this. Reinstall of app did not help.

[image]
[image]
[image]
[image]

smagicld, 2020-11-20:
I’ll increase the limit to 24 cells.

magharees, 2020-11-21:
Great!

For anyone encountering this issue before a fix is merged,  simply use JBLTools to fix

smagicld, 2020-12-20:
fixed

closed 2020-12-20 by smagicld