smagicld/xiaoxiangBMS #122: Purchase Recertification

SV-Juliet, 2020-11-22:
I am using the xiaoxiang BMS Monitor app v1.2.19 on my iPhone and iPad to configure my Overkill Solar 120A 12V BMS to work with my EVE 3.2V 280 Ah aluminum cased prismatic LiFePO4 cells.

I’d like to thank you for creating such a great app! I find it to be very user friendly and it displays very important and useful information. I like the configuration screen and Enable/Disable buttons on the main page.

There is one thing that I’m a little concerned about. After I’ve purchased the app so that I can configure the BMS, the app will prompt me to re-certify the purchase every day or so. Although this is a simple thing to do, it is not an expected behavior as none of my other apps do this. My concern is whether this app purchase recertification requires cell service rather than just reading a certificate on my mobile device. I am using the BMS and app on my boat and will frequently NOT have any cell service. 

Can you please confirm whether the app purchase recertification requires cell service?  If it does, can you please consider this as an enhancement request to NOT do the re-certification so frequently or store a certificate on my device so that the purchase can be re-certified without cell service. It’s important for me to be able to use the enable/disable features and the configuration screen even when I do not have cell service.

ambotme, 2021-02-11:
I came here for this same concern. The ability to not be able to turn off the battery could be very bad in an off grid situation. The purchase history needs to be way more sticky for this to be viable.

smagicld, 2021-02-11:
Normally the purchase status is saved locally, so there is no need of internet connection fo use Pro functions.
but some releases ago there was a bug storing this value locally. 
Just update to the latest version or re-install the app. this will fix the problem.

closed 2021-02-11 by smagicld