smagicld/xiaoxiangBMS #123: Restoring App-Purchase

dadaludwig, 2020-11-23:
Hi, i am having issues restoring my app purchase (€6.99). I have two apple-ID's one of which i used for the purchase (i am not sure anymore which one i used) so i tried both and neither works (attached file shows prompt). Please help/clarify! I urgently need to configure my battery pack (see attached file). Thanks.
[image]
[image]

smagicld, 2020-11-23:
Diese Nachricht kommt von Apple und bedeutet sie konnten kein inApp kauf für das AppleId finden. Da kann ich nicht viel helfen - du musst bei Apple nachfragen.

dadaludwig, 2020-11-23:
ok, verstanden. Danke für die schnelle Antwort. Frage: An wen kann ich mich
da bei Apple wenden? Und können Sie mir sagen, mit welcher Apple-ID ich bei
Ihnen eingekauft habe (oder bekomme ich die Info auch nur von Apple?)? Das
würde mir beim Anmelden helfen damit ich es nicht immer mit zwei ID's
versuchen muss.

VG,
dieter

.. ok