smagicld/xiaoxiangBMS #124: Bat info directly in iOS14 widget

CSYGD, 2020-11-23:
Hi!
It might be gorgeous to have our different battery levels inside the iOS14 widget battery, or inside a dedicate smart BMS widget :) 
All my best [image]

NeariX67, 2020-11-29:
I guess it is not possible to show such data in the default iOS widget as the bluetooth LE device must provide a specific characteristic that includes the current battery status in percent, which the bluetooth module does not provide. The only option would indeed only be a custom widget.

smagicld, 2020-11-30:
To add entries into the Apple Battery widget is not possible. 
but custom widget yes. just made a prototype: 
[image]

it shows the currently connected bms battery status.

smagicld, 2020-12-20:
implemented in V2. drop me your e-Mail if you want to test.

closed 2020-12-20 by smagicld