smagicld/xiaoxiangBMS #127: Writing 0 mA to discharge overcurrent trigger value will set it to 65535 mA

NeariX67, 2021-01-05:
Steps: 
* Set discharge overcurrent trigger value to 0 mA.
* Read BMS data again
* Discharge overccurent trigger reads 65535 mA

smagicld, 2021-01-06:
Thanks for you feedback. the value is changed by the BMS. but 0 mA does not make sense. I'll set the minimal value to 10mA. (1mA is not possible, because resolution of this field is 10mA)

closed 2021-01-06 by smagicld

smagicld, 2021-01-06:
update: look like bug in the app. working on to fix it.

NeariX67, 2021-01-09:
Update 1.2.21 (5354) still has this issue. I can write 0 mA to discharge overcurrent and it won't notify me about a wrong value range and just writes it to the bms. Reading again will set it to 653360 mA

smagicld, 2021-01-09:
can you check this build: https://testflight.apple.com/join/nkMkjap8