smagicld/xiaoxiangBMS #129: MOFSET disabled

Autostar81, 2021-01-15:
Hi,
I put my LIFEPO4 on charge ; after a while, I have an error message : "MOFSET disable"; Anyone could explain me what does it mean and what should I do. RGDS. A "poor" retired man :-)

Skuxxlife, 2021-07-14:
I wonder if the bms has put the battery in a protection mode the mosfet is a circuit that cuts off the battery ground.
Did you ever get it to work?

Autostar81, 2021-07-21:
> Message du 14/07/21 18:44> De : "Skuxxlife" > A : "smagicld/xiaoxiangBMS" > Copie à : "Autostar81" , "Author" > Objet : Re: [smagicld/xiaoxiangBMS] MOFSET disabled (#129)> >

Re,
 
In fact, through the BMS Bluetooth application we can "un-activate" / desactivate the outlet port via the MOFSET.
It is also a safety protection again high voltage or high current.
 
Yes, the battery is able to do its job.
Thanks to have tried to answer my demand.
RGDS