smagicld/xiaoxiangBMS #130: Percent charge changes on re write

markoav8, 2021-01-22:
Frequently by battery shows a very low percent capacity remaining.  If I change any setting (like the switch) and re write to BMS the percent goes way up by 40% or more.  Why?

Banzai1020, 2021-03-21:
Hi Lukas,  Love the app.  One small issue I have is similar the comment above.  I am noticing that in the app (using iOS iPad) that the state of charge (SOC %) does not auto update.  For example, it says 100% when the cells are fully charged (at 3.5 v in my case), but when the cell voltage drops, the SOC is not reflected and still shows 100% which is not correct. (Note, it appears the example photos on the App Store also show the same thing happening (cell voltage of 3.3 v should be about 60 to 80% according to the capacity configuration used, but it is showing 99%).  

I have found that if I do a BMS read, make a minor change to a parameter, then do a BMS write, when I go back to the main  screen the SOC does change to reflect the correct cell voltage.  That is why I suggested refresh problem.

Possible fix for this to come ?
Thanks. Garry B.

smagicld, 2021-03-21:
Hi Garry

The App does only display of the values reported by the BMS Board. the SOC is displayed 1:1 and fully controlled by BMS.
The behaviour you described is not app related. same effect when you do setting changes using the PC software.

In my opinion the only place to fix it is the BMS Firmware which can be done only by manufacturer.

closed 2021-03-21 by smagicld

Banzai1020, 2021-03-21:
Oh darn it.  Thanks though. GB

rocket61, 2021-03-21:
What about having the app do it’s own calculation based on set min and max voltages?

Yuri

smagicld, 2021-03-21:
Hi Yuri

SOC should be calculated by measuring current not voltage.

SOC based on voltage can work only for application with constant current. For applications like eBike or solar power banks this will not work because the load is changing all the time and with the load the voltage drops.

I have experienced this on my solar battery bank where I have finally replaced my SOC voltage indication wit Victron BMV700.

Banzai1020, 2021-03-21:
I also have an AiLi Voltage/Current Meter similar to the Victron.  Just a shame that you install different SOC levels into the BMS, and then it does not use it to calculate %SOC (well it can but just does not keep it updated) :(. GB