smagicld/xiaoxiangBMS #132: Arduino/RaspberryPI API

madmacks59, 2021-02-10:
The software you've built is great and I upgraded to the pro version in support of your effort. Thank you. Is it possible that you could generate an API library that could be used on the Arduino platform, NodeMCU, or the Raspberry Pi platform. I live off grid and have a total of 6 different solar/battery systems that I monitor. At each of these locations I have a small NodeMCU based system that monitors voltages, temperatures, and various other states. These values are then logged to a remote MariaDB/MySQL DB that tracks history. I then have a web server running on a Raspberry Pi that gives me a one page view of the state of all my battery systems. It would be AWSOME if I could use the NodeMCE (which supports BlueTooth) to read some of the states of the BMS via a library. You obviously have divined how to do that with this project, and maybe you could throw out a "skinny" library that would allow people to read BMS state values.

armstrongsolona, 2021-03-07:
I would like to second this request. My friend and I are setting up a Raspberry Pi with Home Assistant and an ESP32. We just built a LiFePO4 with the xioxiang BMS and would love to be able to control it through Home Assistant with an API! Thank you.

spacesuitdiver, 2021-06-03:
Same, I came here expecting some source code to reverse engineer to ESP32 via the UART port. I'll post here my findings if you all share your's. This is as far as I've gotten https://wnsnty.xyz/entry/jbd-xiaoyang-smart-bluetooth-bms-information

Raspb-henk, 2022-08-02:
Hi
For this kind of  librarie look at:
https://github.com/FurTrader/Overkill-Solar-BMS_2-Arduino-Library
or
https://overkillsolar.com/open-source-projects/
From there you can find several treads on this subject. I have used the software on a NodeMCU and modified it so it sends the data to a MQTT server. I then use Node-red to put  it into a Influxdatabae on my Raspberry PI and present it with Grafana.