smagicld/xiaoxiangBMS #134: MOSFET software locked

Pixel-est, 2021-02-22:
I got an alarm saying mosfet software locked now my e-bike won’t charge but there is current going through the charger and the battery can someone tell me how to fix this issue

smagicld, 2021-02-22:
This means the battery was switched off (by software). You can press the Batt On button on the main screen to switch it on.

Pixel-est, 2021-02-23:
I can’t do that because I don’t own the pro version. I’ve been toying
around with it while it was plugged in although i don’t think it’s charging
because the fan isn’t working anyway I gently pulled on the throttle and
the fan started spinning for the duration I pulled the throttle probably
time to get a new charger

closed 2021-06-13 by smagicld