smagicld/xiaoxiangBMS #139: iPad app won’t read bms

itsnotlevel1, 2021-04-07:
It seems like all the other functions work, if I try to do configure it won’t read the actual BMS any thoughts?

smagicld, 2021-04-07:
you need to provide more details.
- App version
- BMS type (hardware/model)
- please also provide BMS logs. Go into Info Screen. Tap on the Build number you should get an confirmation message. Go back to BMS, try to read settings. Afterwords go back into Info Screen and tap build number again. send the email.

itsnotlevel1, 2021-04-11:
Sorry for the late reply. Work....blah. BMS Name: JBD-SP04S020-L4S-120A-B-U-K BMS Model: 12v 120A BMS SW version: 2.0 BMS Production date 01/15/2021 Purchased from Overkill Solar. 
[debug.zip](https://github.com/smagicld/xiaoxiangBMS/files/6292671/debug.zip)

itsnotlevel1, 2021-04-11:
[debug.zip](https://github.com/smagicld/xiaoxiangBMS/files/6292780/debug.zip) The first upload may not have had anything in it.

smagicld, 2021-04-12:
Thanks. I checked the logs, could be related to the new BMS Firmware. 

I've made a test build where I would like you to try it. please provide me your e-mail so I can add you to testflight test group.

The new Version is: 1.2.23, Build: 5405 (just in case you see in TestFlight other version - it takes some time by apple to release it)

closed 2021-06-13 by smagicld