smagicld/xiaoxiangBMS #141: Write button missing for voltage calibration.

jimott, 2021-04-27:
I can't calibrate the bms voltages because there is no write button.  I am using the latest apple pro download and IOS 14.5.  Same problem with the previous IOS version.  Iphone 11 pro.
[image]

smagicld, 2021-04-27:
Hi. 

Just tried to reproduce on iPhone 12 Pro and 14.5 - works ok.

Can you please go into Settings -> Display  and reduce the Font size  and switch off bold text?

jimott, 2021-04-27:
I changed to the smallest font and switched off bold text.  It Works!   Problem solved...Thanks

Nighty666, 2021-11-11:
Can you click on „Online Manual“? I DL the latest iPhone version, but nothing opens if I click on it.  I want to know if I should do the calibration at 0 Amp. or during I charge with 10 Amp‘s for example.

closed 2022-07-21 by smagicld