smagicld/xiaoxiangBMS #142: Problem with scrolling the display

Romanidse, 2021-05-02:
I have an iphone 8 plus and IOS 14.4 
I have around 200 BMS and I can not scroll down to find the right bms.

smagicld, 2021-05-02:
Please submit some screenshots.
Why do you see 200 BMS? do you have so many of them at your place?
Which App version are you using?

Romanidse, 2021-05-03:
It's better to show it in a video, I can do it tomorrow, you will see I can not scroll down. Yes we have so many BMS (ebike batteries). I have the app version 1.2.22 build 5391

Romanidse, 2021-05-05:
I made two video and upload them to wetransfer. here you can download them https://we.tl/t-lgt8LDOgYD

Romanidse, 2021-05-09:
did you check the videos?

smagicld, 2021-05-10:
Yes. I have an idea. Drop me your email - once I have an beta release I’ll ask you to do a test. .

Romanidse, 2021-05-10:
great thank you. my email is roman@chevrom.de

smagicld, 2021-06-13:
HI Roman.

The scrolling problem is related to backgroud scan refresh.
I'll implement an toggle on main screen to trigger on/off the scan. so you'll be get the list of BMS's and then disable auto scan. this should fix the problem.
should be released in 2-3 days.

smagicld, 2021-06-13:
Hi Roman

I've implemented toggle to pause device scan so you should be able to scroll over you devices. I'll release now TestFlight. The Build Number is: 5489

Let me know if you're ok with that!

[image]

Romanidse, 2021-06-16:
Hi, thanks, I tested and its working now.

Romanidse, 2021-06-16:
> Cool. So I’ll release it today.
>
> Von meinem iPhone gesendet

ok thanks

closed 2022-07-21 by smagicld