smagicld/xiaoxiangBMS #143: My Beta has expired

TheDodgersOnline, 2021-05-02:
Hi - can you enable my access again please? eMail is rdodd9@icloud.com

TheDodgersOnline, 2021-05-09:
Please?

TheDodgersOnline, 2021-05-30:
Seriously? 27 days and counting. How can I get this re activated.

Magsybaby, 2021-06-04:
I would be really keen to get back in also, the datalogging / export feature was amazing :)

TheDodgersOnline, 2021-06-05:
Yeah - been waiting a while but no response. Not sure why?

smagicld, 2021-06-05:
Hi

Since 2 weeks I’m trying to release it as new App in appstore. Curently Apple is rejecting because of admin stuff (link gdpr or dedicated contact site) that I have to create first. it takes time…..

But for the meantime I’ll submit today evening new beta for testflight until apple releases the app.

Magsybaby, 2021-06-07:
Thank you for that, it’s back :) :)

smagicld, 2021-06-13:
beta is refreshed. working now to have it in app and macStore released but apple is making some problems...

closed 2021-06-13 by smagicld