smagicld/xiaoxiangBMS #145: Prevent charging after short discharge

leichti, 2021-05-08:
Hello,

the BMS in its default state allows charging after a short discharge, even if the cell voltages are above the release value. As far as i remember, the JBDTools had the option to disable this behavior (So that the BMS only allows charging after the cell voltages fall below the release value). Is it possible to add this option in this fantastic app? Would be great for my project.

smagicld, 2021-06-13:
yes, I can check this. Can you post me the screenshot / parameter name from JDB Tools? - just to be sure we're talking about the same,

rayw45, 2021-06-21:
Is this the same problem I am having with this otherwise superb app I wonder?  I get the cell under voltage protection step in as my inverter does not cut off before one of my less good cells drops below the 2.5 V under voltage threshold.  Problem is although I have the app set to have charging port on when battery off, when the under voltage protection steps in my charger (or later when I get solar panels will be my MPPT) cannot start recharging the batteries as is of course then urgently needed.  Surely the under voltage protection should only shut off the discharge port not the charging port otherwise the batteries will remain discharged permanently without what shoujld be an unneeded manual intervention ???  Yes I know that you can get BMS devices with a separate charging port but I just cannot find any of these for sale online and thought with the allowing of charging port on when battery is off would allow this to happen with these otherwise superb overkill type common port devices and your amazing iphone app for which I have purchased the Pro version of course.

rayw45, 2021-07-09:
Anybody got ananswer to my above post please ??