smagicld/xiaoxiangBMS #146: Voltage bar graph shows wrong values

microfarad-de, 2021-05-12:
Dear Developer,

I'm a Pro user of your great BMS app and would like to report the following bug.

After flipping the iPhone into landscape mode and back to portrait, the bar graph that visualizes the individual cell voltages starts showing inconsistent levels for cells having the same voltage (see screenshot). 

Kindly investigate this issue. Otherwise a very good app. Thank you!

Best Regards,
Karim Hraibi

[image]

microfarad-de, 2021-05-12:
Forgot to mention my device model: im using iPhone 12 Pro

microfarad-de, 2021-05-14:
OS version: iOS 14.4.2

smagicld, 2021-05-14:
Thanks, will check it.

Jaco-TTT, 2021-12-16:
I have 18 cells and over time the visual display of the cells goes completely out of sync in landscape on an iPad. Worked on 17 cells though. 

On the left, where the Balancing appears, the blue line per cell goes nearly all the way to the right, over the word Balancing.

Turning the iPad back to portrait, the visual presentation is "fixed" again, then one can turn it back to landscape mode.

IPad is one all the time.

Ps. yes, 18 cells on a 48v bank.