smagicld/xiaoxiangBMS #148: Waiting for Data

Solavox, 2021-06-03:
Using iOS version of app 1.2.22 build 5391.

I used the app to save a BMS config of BMS A.

I then connected to BMS B, opened the saved config of BMS A and wrote it to BMS B. I got an error writing on the BMS name field. 

I disconnected and then tried to reconnect to BMS B. It connects and has the ‘waiting for data’ statement with no option to do anything except go back to the BMS connection screen.

The battery is still charging and discharging.

I haven’t removed the Bluetooth module or disconnected the BMS at this stage.

smagicld, 2021-06-03:
Need some more details. please send me the debug logs:

1. Go to about screen and tap on the version / build number once:

2. it should be confirmed:

3. go to main screen. try to connect to your BMS. wait 2 minutes (to collect some data)
4. go back to about screen and tap on version / build number
5. send me the logs over email

thanks!

Solavox, 2021-06-03:
Logs sent just now - thanks!

smagicld, 2021-06-03:
you have not executed step 3. I do not see any BMS data. 

after enabling logging try to connect to bms.

Solavox, 2021-06-03:
I’ll try again - hold on

smagicld, 2021-06-03:
going now to bed (1:00 am).. will take a look 2morrow

Solavox, 2021-06-03:
Ok - sleep well. I’ve rerun it - file size is coming out the same.

attached are some screenshots (I’m connecting to U1timatetron).

[image]
[image]

smagicld, 2021-06-04:
That’s wired…

In the reported Log from you I do not find any Bluetooth activity, just one line:
[ℹ️] [2021-06-04 10:51:52.9190] submitRecordsButton(_:)[110]: Logging started by user

Looks like you would start debug and then stop it without trying to connect into bms. Can you do some screen recording (from enabling debug, connecting to bms till debug off).

But before you do this: have you tried to reinstall the app?
And: the BMS you’re trying to connect: are this Xiaoxiang BMS? (which model?)

(to compare, my test from today)
[ℹ️] [2021-06-04 07:27:57.5840] submitRecordsButton(_:)[110]: Logging started by user
[💬] [2021-06-04 07:28:02.4040] peripheral(_:didDiscoverServices:)[1165]: didDiscoverServices [uuid: 19A6F200-638F-2602-E794-280A3CD7FDF1]: <CBService: 0x280606cc0, isPrimary = YES, UUID = FF00>
[💬] [2021-06-04 07:28:02.4050] peripheral(_:didDiscoverServices:)[1165]: didDiscoverServices [uuid: 19A6F200-638F-2602-E794-280A3CD7FDF1]: <CBService: 0x280604180, isPrimary = YES, UUID = 1912>
[💬] [2021-06-04 07:28:02.4640] peripheral(_:didDiscoverCharacteristicsFor:error:)[1186]: didDiscoverCharacteristicsFor [uuid: 19A6F200-638F-2602-E794-280A3CD7FDF1, service: FF00]: <CBCharacteristic: 0x28377b0c0, UUID = FF01, properties = 0x12, value = {length = 15, bytes = 0xdd0400080d660d790d760d50fe1f77}, notifying = NO>
[💬] [2021-06-04 07:28:02.4650] peripheral(_:didDiscoverCharacteristicsFor:error:)[1198]: Rx and notify char.: FF01
[💬] [2021-06-04 07:28:02.4650] peripheral(_:didDiscoverCharacteristicsFor:error:)[1204]: Rx char.: FF01
[💬] [2021-06-04 07:28:02.4660] peripheral(_:didDiscoverCharacteristicsFor:error:)[1186]: didDiscoverCharacteristicsFor [uuid: 19A6F200-638F-2602-E794-280A3CD7FDF1, service: FF00]: <CBCharacteristic: 0x28377aa00, UUID = FF02, properties = 0xE, value = (null), notifying = NO>
2021-06-04 08:49:52.700413+0200 XiaoxiangBMS[74074:845279] [] nw_protocol_get_quic_image_block_invoke dlopen libquic failed
[💬] [2021-06-04 07:28:02.4670] peripheral(_:didDiscoverCharacteristicsFor:error:)[1180]: uuid 1912 not found in xiaoxangServiceUUID_V1/2
[ℹ️] [2021-06-04 07:28:02.6280] peripheral(_:didUpdateValueFor:error:)[1108]: received bytes from BMS: DD 04 00 08 0D 66 0D 79 0D 76 0D 50 FE 1F 77 00 00 00 00 00
[ℹ️] [2021-06-04 07:28:02.6370] decodeSingleBinaryAndUpdateObject(binaryData:)[235]: --> decodeSingleBinaryAndUpdateObject DECODED .CELLS_VOLTAGE

smagicld, 2021-06-13:
no user feedback. closing.

closed 2021-06-13 by smagicld