smagicld/xiaoxiangBMS #149: BMS Funktioniert nicht richtig

kai36ks, 2021-06-08:
Hallo ich habe bei Ali Express das JBD Smart BMS 4S 12v 150 A gekauft und in mein Wohnmobil eingebaut. Zu dieser Zeit nutzte ich ein Android Handy mit der genannten App, alles funktioniert prima. Jetzt habe ich ein Apple i Phone xr mit der Xiaoxiang BMS pro App diese funktioniert soweit, außer dass ich das BMS nicht auslesen kann. Batterie Daten auf der ersten Seite werden angezeigt. Wenn ich zu den Einstellungen wechsle und dann auf BMS auslesen gehe dauert es einen Moment und ich bekomme die Fehlermeldung "  Fehler , keine Bestätigung vom BMS Versuche noch einmal". Das auslesen mit der Android App dagegen klappt einwandfrei.  was mache ich falsch? Woran kann es liegen? Mit freundlichen Grüßen Kai Frühwald 
[image]
[image]

smagicld, 2021-06-08:
Hallo Kai

Brauche mehr Details.

Wie ist genau die BMS Bezeichnung? (Modell). Hast du noch den Ali link?
Welche Android App genau verwendest du?

Das Problem ist, das der Hersteller mit der Zeit Firmware Anpassungen vornimmt. Diese müssen dann nachprogrammiert werden.

smagicld, 2021-06-13:
in Arbeit. muss das gleiche BMS bestellen...

p-f-w, 2021-06-24:
ich würde sagen, das Passwort des BMS ist gesetzt, die Android Apps fragen es ab (außer Liontron/Solmax, dort glaub ich aber auch nicht, daß der Befehl überhaupt richtig ist rw-Mode A5 anstatt 5A)

Falls beim Start das "Pairing" mit dem BMS nicht funktioniert, fragen die Android Apps von "Carplounge BMS" 3.1.1020 aus dem Play Store bzw die Xiaoxiang-App Version 3.1.1026 das Passwort ab.
Verwenden aber anfangs verschiedene Varianten des "Pairing", CarploungeBMS sendet das Pairingcommand nach zwei Basisinfo-Abfrage, Xiaoxiang sendet erst eine Zellinfo-Abfrage, dann eine Basisinfo-Abfrage, glaube nicht, daß das entscheidend ist

XIAOXANG             3.1.1026  -> DD5A060706000000000000FFED77    (keine Abfrage, falls nicht gesetzt oder 000000)
DD Startbyte
5A rwmode
06 command
07 Länge (Länge Längenbyte + Passwortlänge(immer 6), die App unterstützt keine andere Länge, Passwortänderung funktioniert auch nur mit 6stelligen altem und neuem Passwort))
06 Länge
00 1. Stelle Passwort 00 für 0, 01 für 1 ... 09 für 9
00 2. Stelle
00 ...
00
00
00 6. Stelle
FF Checksumme 1. Byte
ED Checksumme 2. Byte
77 Stopbyte
Carplounge BMS    3.1.1020  -> DD5A060100FFF977                            (keine Abfrage, falls Passwort nicht gesetzt)

Solmax/Liontron(alt) 3.1.004   -> DDA50600FFFA77                                (ob das überhaupt funktionieren konnte, k.A. s.o., die fragen aber auch nicht nach )

Antwort DD060000000077 , falls korrekt
Antwort DD068000FF8077 , falls nicht korrekt, falls ich mich recht erinnere

setzen des Passwortes auf 000000 per App war in meinen Tests nicht dasselbe wie Löschen des Passwortes per JBD Tool (JBD Tool ältere Version konnte auch dann die Parameter nicht auslesen)

Passwortänderung:
DD5A070D0C'sechsstelligesaltesPasswort''sechsstelligesneuesPasswort''Checksummenbytes'77

Passwortchange-Response, wenn erfolgreich:
DD070000000077