smagicld/xiaoxiangBMS #150: Log watt-hours (wh) and amp-hours (ah) coming “in” and “out”

TacticalHubs, 2021-06-23:
Hello,

I live your BMS and the BMS app. Since the BMS seems to be already recording all the right data any way (voltage x amps x time), it would be most helpful if the app could display and log the sessions total watt-hours (wh) and amp-hours (ah) that have gone “in” to the battery when charging, as well as “out” of the battery when discharging to a load. Is that something you could release into an update soon?  Would greatly reduce the amount of meters needed if I can get that key information in the BMS app. One place to log in and see all critical data!

Thank you again!

Joe

TacticalHubs, 2021-06-26:
Hallo @smagicld 

Ich liebe deine BMS-App. Da das BMS anscheinend bereits alle richtigen Daten (Spannung x Ampere x Zeit) aufzeichnet, wäre es am hilfreichsten, wenn die App die gesamten Wattstunden (wh) und Amperestunden (ah) der Sitzungen anzeigen und protokollieren könnte. die beim Laden „in“ in die Batterie gegangen sind, sowie „aus“ der Batterie beim Entladen zu einer Last. Könnten Sie das bald in einem Update veröffentlichen? Würde die Anzahl der benötigten Zähler erheblich reduzieren, wenn ich diese wichtigen Informationen in der BMS-App abrufen kann. Ein Ort, um sich anzumelden und alle wichtigen Daten zu sehen!

Danke nochmal!

Joe

smagicld, 2021-06-26:
Hey Joe
Diese Funktion habe ich bewusst noch nicht umgesetzt weil:
- das BMS speichert diese Daten nicht
- sollte ich die Werte auf iOS messen, wo wird nach Verbindungsunterbruch falsch gerechtet (weil Daten haben gefehlt).
Probiere aber mal die neue Version:
https://testflight.apple.com/join/HIiG2IEr

hier kannst du Logging verwenden. Anhand der Daten kannst es dir ausrechnen.

Gruess
/lukas

TacticalHubs, 2021-06-26:
@smagicld 

Danke. Ich habe gerade die neue Version in TestFlight heruntergeladen. Ich werde damit spielen und euch Feedback geben.

TacticalHubs, 2021-06-26:
@smagicld 

Hallo, Ich habe vier Fragen, die ich im Handbuch nicht alleine beantworten konnte, also schätze deine Hilfe: 

1) Was ist der Unterschied zwischen "Gesamtbatteriekapazität" und "Gesamtzykluskapazität"? Ich habe 8x 3,2 V-Zellen mit jeweils 280 Ah in Reihe geschaltet, um 1x 24 V-Akku zu bilden. Ich gehe davon aus, dass die "Gesamtbatteriekapazität" 280 Ah beträgt, aber warum ein anderes Feld "Gesamtzykluskapazität" haben? Kann ich das einschränken, wenn ich möchte (Beispiel: Wenn ich nur von 90% SOC auf 20% SOC gehen möchte, würde ich die "Total Cycle Capacity" auf 196ah ändern (das sind 70% von 280ah))? 

[image]

2) Wozu dienen die NTC-Einstellungen? Ich sehe 8x Kästchen, die aktiviert oder deaktiviert werden können. Ich bin mir nicht sicher was ich mit diesen Boxen machen soll. 

[image]

3) Wie kalibriere ich den "Strom"? 

[image]

4) Wie kalibriere ich die "Spannung"? 

[image]

Ich habe ein 0-30v 0-10a variables Ladegerät sowie ein Delta-q IC1200 24v 50a Ladegerät. Vielen Dank für Ihre Hilfe bei der Kalibrierung von "Strom" und "Spannung" in den App-Einstellungen.

 Joe

smagicld, 2021-06-26:
HI Joe
1. Die App ist nur eine UI auf die Params. Darum am besten immer beim BMS Hersteller fragen.
Gesamtkapazität - ist klar.
cycle Kapazität - was wird entladen zwischen full und cut-off.

2. An den BMS kann man bis zu 8 Temp. Sensoren anschliessen (NTC).

3. Zuerst Idle Strom. Dann Last anschliessen, Strom messen und den gemessenen Wert eingeben. das gleiche beim Laden.

4. Spannung von jeder Zelle messen und den Wert eingeben.

Wichtig: Für die Kalibrierung benötigt man ein Kalibriertes Messgerät (diese haben einen entsprechenden Zertifikat. z.B. Fluke Multimeter)! Arbeiten mit günstigen China Multimeter ergibt kein Sinn da der selber nicht kalibriert ist.

Gruss
/lukas

TacticalHubs, 2021-06-26:
@smagicld 

Danke nochmal!