smagicld/xiaoxiangBMS #152: Something isn’t right…

FatDaddyG, 2021-07-01:
[image]

smagicld, 2021-07-01:
can you specify what exactly you mean?

FatDaddyG, 2021-07-01:
Yes.. I have 6 of these and this particular BMS is set up exactly as the others.  Shows 70% SOC but voltages are at max.  Other 6 batteries show ~98% all connected to same buss and charging for months.  Could be a hardware issue I guess?

smagicld, 2021-07-01:
All values are displayed 1:1 as reported by BMS. This would mean that the one BMS has wrong settings (100% , 80% 60%, 40% voltages...)

p-f-w, 2021-07-01:
may be that the 7. bms is a other(newer(?)) one? Newer bms has settings for 10%,30%,50%,70%,90%,100%, that can not be set via android app or other apps are not adapted, only with newest(?) JBD Tools

please take a look at https://gitlab.com/bms-tools/bms-tools/-/issues/30 (latest entry there and a screenshot posted before by other one)

closed 2022-07-21 by smagicld