smagicld/xiaoxiangBMS #158: BMS giving low Amp when I discharge

JorgoChr, 2021-08-29:
Hello,

I have made an electric bike for my personal use. At the beginning my bike drove as I accepted (high amp on discharge). After I drove it a bit it started to give me only 0,5A on discharge. Can anyone tell me maybe where the problem is exactly? I have changed the settings at the beginning without saving the default settings. So I don't remember them anymore. 

I have made an 14S battery with 48V and 20Ah but I am only able to use like 5,5Ah. I would be more than happy if anybody know how to fix the problem.

[image]

AndersVikingSvensson, 2021-12-02:
The cells are not fully charged ! Shoul read 4.17V each cellgrup

RCinFLA, 2022-02-05:
Also check if BMS is set for LiPo or LiFePO4 battery type voltage range.  For 48v systems, 14s battery packs are usually LiPo (4.2-3.5v per cell range).  16s battery packs are usually LiFePO4 (3.65-2.5v per cell range).

Your voltage readings look like discharged LiFePO4 cells, or over-discharged LiPo cells.

They may also need balancing.

closed 2022-07-21 by smagicld