smagicld/xiaoxiangBMS #159: « In apps purchase » always disabled

Antoine-Guillemin, 2021-09-03:
I am using the app to turn On and Off my battery.
Main problem is that I have to restore every time the « in apps purchase » which require an internet connexion.
This mean that even after buying de 7€ Pro version, I cannot use my e-bike battery if I park myself in an underground parking or a location without internet connexion.
Could you update the App to avoid this verification at each use of it. 
Thank you

smagicld, 2021-09-03:
Hi

Just delete the app, install newly form appstore and restore purchases.

then it should be always pro without internet connection.

let me know about the result.

cheers
/lukas

Antoine-Guillemin, 2021-09-03:
I tried and I still have the same issue (On the V1 version). 
When I open the app, I have to restore the purchase at every opening.

smagicld, 2021-09-03:
which version of the App are you using?

closed 2022-07-21 by smagicld