smagicld/xiaoxiangBMS #160: BMS working but wrong SOC

jbowen383, 2021-09-09:
I connected the BMS with the 280Ah at 50%charge and now the app says 100%charged at 100Ah and in small numbers beneath it says 99.51% charged and continues to grow the individual cells appropriately and I’m wondering if there is a start over button when it reaches full capacity?  Thanks if you help and thanks for just reading this.

smagicld, 2021-09-09:
The app displays only values reported by bms. 
It looks like you have changed some bms settings or did not configured it at all. For support ask you battery reseller. The settings must match used cells.

jbowen383, 2021-09-09:
Thank you for your response. Very helpful.

RCinFLA, 2022-02-05:
You have to set battery AH rating in BMS and charge voltage to reset Columb counter to 100% full.

Some 'smart' units find this info by themselves but require training by full range charge/discharge cycling of battery capacity.

closed 2022-07-21 by smagicld