smagicld/xiaoxiangBMS #161: How to change battery mAh settings/ configure

Lostmarbles87, 2021-09-22:
I currently have a 4c 12v totaling up to 280Ah but the bms is only saying 100Ah on display.

smagicld, 2021-09-22:
100AH is the default of Xiaoxiang bms. 

Go into settings and adjust total capacity (it is in mAh).

Lostmarbles87, 2021-09-23:
Thank you!
Which values need to be manually input and which values will the bms figure on its own?

Comrad21, 2022-01-07:
Hi, my pack is 720 Ah but app won’t let me set it that high. Is there other options or work around?

matthewrichesmarine, 2023-02-03:
> 

Comrad - Did you find a solution to this? I need to enter 840Ah which is 840000mAh but it only allows a maximum value of 355350mAh.

Thanks!

smagicld, 2023-02-03:
Hi. This lis BMS limitation. it allows to write max 65535 * 10 mAh. can not be fixed in the app.

closed 2023-02-03 by smagicld