smagicld/xiaoxiangBMS #162: Reset the BMS Name after set PIN1/PIN2

Jethro1000, 2021-09-28:
After i set PIN1/PIN2 now the BMS is exposing the PINs via BT. How can i set back the BMS name and remove the PINs. 
Special for this, i have no UART adapter, a USB to RS232 Dongle or a BMS software for my BMS

smagicld, 2021-09-28:
Just go into settings and set the pin toggle off and set your custom name.

Jethro1000, 2021-09-28:
And this didn't work
Step by Step:
Toggle Pin Off
Set Custom Name
Write to BMS

It didn't change the name and old is exposed Pxxxx\Pxxxx (also checked with BLE Scanner)

smagicld, 2021-09-30:
Ok, I understand now. You want to change the Bluetooth advertisement name. 

The Bluetooth module can not be controlled by bms. If you want to have your own Bluetooth name you have to replace the dongle with an that can be configured (hc05/06/10). 

Then you can set your custom name with an usb-ra232 (ttl) Interface.

Jethro1000, 2021-10-01:
I think your answer isn't correct. With the PIN Setting the advertisement name was overwritten by your software. The only thing what i want, is to restore the old name

smagicld, 2021-10-01:
Believe me - my software can not change the advertised Bluetooth name because it is not supported / there is no code to do that. The Bluetooth dongle is separate peace of hardware.

f-io, 2022-08-17:
OT: Any known source for a fw file? I've bought mine from lithiumbatterypcb.com, it is a SP15S008C-15S BMS, FW 19.