smagicld/xiaoxiangBMS #163: Lose of balance during charge at high percentage

Lostmarbles87, 2021-11-24:
While charging my 4c 280ah 12v LiFePO4 battery I loose balance between my 1,2 and 3,4 cells. The problem seems most noticeable at higher percentages (80% and above) and when charging off my 40Amp DC-DC charger. My 1,2 cells will get to 3.65v limit triggering voltage protection while the 3,4 cells will usually sit around ~3.34 range. 
note: will happen when all cells are balanced prior to charging. 
[image]

finoshiny, 2021-11-28:
The same problem for me

RCinFLA, 2022-02-05:
Cells are either out of balance or poorly matched.

smagicld, 2022-07-21:
This is not an App issue - it is the FW of BMS. please ask manufacturer for support.

closed 2022-07-21 by smagicld