smagicld/xiaoxiangBMS #164: iOS version 1.2.25 bug

alexgrant92, 2021-12-13:
Hello. Since I updated the app I have had issues viewing my voltage levels through the app. Voltage number on the left appears correct but voltage bars on the right fluctuate constantly and are not accurate compared to the voltage  displayed on the left. Recently happened after updating the app. 

I have deleted app and reinstalled with no help. 

https://user-images.githubusercontent.com/64763428/145891936-43d7416c-0aa1-45d6-b0a3-9a20e44916a4.MP4

Jaco-TTT, 2021-12-16:
Jip, I've seen the same on a iPad. 

Turning it upright, then back to sideways, helps to alleviate the issue.

smagicld, 2021-12-16:
Yes, this is know bug.

Are you using the app to monitor you BMS constantly?
Try the new Version: BMS Tool https://apps.apple.com/at/app/bms-tool/id1511343379 

there you have the view fixed (SwiftUI) and in the app settings you have the option to keep the display always on! (I’m using it to constantly monitor my batteries in camping car on an iPad)

Jaco-TTT, 2021-12-16:
> Yes, this is know bug. Are you using the app to monitor you BMS constantly? Try the new Version: BMS Tool https://apps.apple.com/at/app/bms-tool/id1511343379 there you have the view fixed (SwiftUI) [...]

Perfect thank you!

Use this version for constant view and the paid version for programming the BMS, correct?

alexgrant92, 2021-12-16:
Bms tool works without issue but I’m unable to edit bms configuration but I will do as suggested and use both apps for different uses. Thanks guys

-Alex Grant

closed 2022-07-21 by smagicld