smagicld/xiaoxiangBMS #165: BMS Pin iOS 15.2

DarkDogOnTour, 2021-12-17:
Bisher hat die Bezahlversion klaglos ohne Pineingabe, resp. Abfrage funktioniert, seit dem Update auf 1.2.25 kommt der Screen mit der Abfrage, wenn ich in die Konfig.  will. Hab wissentlich nie einen Pin gesetzt, und die 000000 klappt auch nicht.

Solmax/Liontron 100Ah24V, 06202100065 … Juni 2020?

Gibts eine Möglichkeit, auf die vorige App-Version zurückzusetzen?

LG Bernd

smagicld, 2021-12-18:
Hallo Bernd

Liontron war nie offiziell unterstützt in der XiaoxiangBMS app. 
Als workaround nutze bitte das BMS Tool https://apps.apple.com/ch/app/bms-tool/id1511343379  - ist neuer. 

Für XiaoXiang werde ich eine Option einbauen wo ich Liontron zu erkennen versuche und die Hardware PIN Abfrage nicht mehr zeige. 

Ps: kannst du mir die debug logs senden?

1. Gehe in info Screen. 
2. tappe auf die Build Version. Eine Bestätigung soll kommen.
3. verbinde dich mit dem BMS. Versuche das HW PIN einzugeben. 
4. gehe zurück ins info Screen und tappe wieder auf die Build Version.
5. sende mir die email. 

Danke.

DarkDogOnTour, 2021-12-18:
Das Logfile ist abgeschickt.

DarkDogOnTour, 2021-12-18:
per mail klappts nicht...

[debug.zip](https://github.com/smagicld/xiaoxiangBMS/files/7739382/debug.zip)
.

DarkDogOnTour, 2021-12-18:
V 1.2.26
Build 5589