smagicld/xiaoxiangBMS #167: Inconsistent Charge Display

3morex, 2022-01-05:
I am using the latest version of the app on a new iphone se.

When charging via solar or the van engine the app displays the charging data correctly.

When charging via the mains charger the app instantly displays a charge of 100% - 400ah of 400ah and this reading remains even after charging stops.  This makes it impossible to actually know the SOC of the battery.

smagicld, 2022-01-05:
Hi

The problem is not related with the app. The app just displays what is reported by BMS board. 

Check your bms connection, if you use correctly charge / discharge ports. If this will not help - ask the bms manufacturer.

3morex, 2022-01-05:
Thank you for the quick response. I will contact my battery supplier.

closed 2022-07-21 by smagicld