smagicld/xiaoxiangBMS #168: Switching charge and discharge port on/off.

itchako, 2022-01-09:
Would be great if switch buttons could be added to switch charge und discharge MOS-Fet‘s on/off.

kind regards
itchako

smagicld, 2022-01-09:
Can you explain in detail what you mean? What is your use case?

There is already config option to switch off the charge together with discharge mosfet or just discharge mosfet only.

itchako, 2022-01-09:
Have several batteries behind a charge booster which can provide 30A max. But if one battery is 100% loaded, the booster still shows around 10a current going in.

My Intension is to switch off the charging Mos-fet of the fully loaded battery to add these 10a to the loading current for the other remaining batteries to optimize charging time.

In Version 1 of the tool there is such a switch button but that is stopping charge and discharge Mos-fets together. At least I did not find a possibility to Stopp only the charging Mos-fet. Means I am looking for independent Buttons for charge and discharge Ports.

Hope that helps.

Kr
Jakob

*** Semper Paratus ***

smagicld, 2022-01-09:
Hi Jakob

you’re the first with such use case.

so what would you say about toggle buttons under the arrows to enable / disable the input/output MOSFET?

itchako, 2022-01-09:
Sounds great!
Thanks in advance. 

Kr
J

smagicld, 2022-01-22:
Hi Jakob

Please send me your email to [xiaoxiang at e77code.com] so I can add you into the beta testing (TestFlight).
the toggles are there (Version: 1.2.28, Build: 5708)

once I got your e-mail and apple released the build you'll get an invitation email.
[image]

RCinFLA, 2022-02-05:
Some BMS's allow this to be done but it has some negative consequences.

There are two back to back MOSFET's to provide true bi-directional switching.  Each of the back to back MOSFET's have a body diode that will allow current to pass in opposite direction, with a diode voltage drop.

Turning off the MOSFET responsible for passing charge or discharge will result in other function having a diode drop which can cause a lot of self heating if too much current is drawn through turned off MOSFET body diode.

Unit's that have this 'charging' or 'discharging' on-off capability also monitor the pass current and override the user manual setting, turning back on the user deactivated MOSFET is too much current is drawn.