smagicld/xiaoxiangBMS #172: Eingabe der Hardware PIN für nicht auf Konfigseite

tomac01, 2022-02-13:
Hallo,
nach dem ich in meinem JBD-SP04S020L4S-100A den Hardware-Pin gesetzt habe, konnte ich die Konfiguration weiterhin aufrufen. Leider musste ich die App neu installieren. Wenn ich die Konfiguration nun aufrufen möchte werde ich nach der PIN gefragt. Nach der Eingabe lande ich aber nicht in der Konfiguration sondern ohne Fehler wieder im Dashboard.

Ich bitte um Hilfe!

Vielen Dank!

smagicld, 2022-02-14:
Hallo

Ja, sieht nach einem Bug aus. 
Wir arbeiten daran!

smagicld, 2022-02-14:
Hi
Bug wurde gefixed, Version 1.2.29 sollte morgen im AppStore erscheinen.
Danke fürs melden!

closed 2022-02-14 by smagicld

tomac01, 2022-02-14:
Ich gebe gerne Feedback, ob es dann funktioniert. Vielen Dank für den schnellen Fix!