smagicld/xiaoxiangBMS #174: No connection to iphone with XiaoxiangBMS (Pro) , but BMS tool works

ClowZee, 2022-02-20:
Hello,
I´m not able to connect my iphone with the Xiaoxiang app, whether its Pro or not. The blue LED on the BT module remains off.
Checked with "BMS tool" and it works pretty good. 
What am i doing wrong or is it a bug?

smagicld, 2022-02-20:
If one app is already connected to the BMS then it does not broadcast. you have to disconnect from other Apps to make it visible again.

closed 2022-02-20 by smagicld

ClowZee, 2022-02-20:
Thanks, I did several times. Deleted both and installed single app only. Same problem...

smagicld, 2022-02-20:
ten you have some other devices which is connecting to the bms.
which version of the apps are you using?
share screenshots from BMS Tool and Xiaoxiang BMS first screen (where devices are listed)

ClowZee, 2022-02-20:
Checked the settings of both apps and compared:
Inexplicably, Bluetooth was deactivated for the Xi.App and activated for the BMS tools in the iOS rights assignment during installation. Strange. But solved!
THX