smagicld/xiaoxiangBMS #175: Hardware pin

lullabyforkful, 2022-02-26:
Tapping the hardware pin button doesn’t prompt me to select a pin code. It disconnects me from my device, and stops the Xiaoxiang app from finding my device . I can no longer find or access my device through Bluetooth. With or without a pin. Any advise? Does it disable Bluetooth access?

Thank you,

smagicld, 2022-02-26:
Please attach video. 
Which model of BMS are using?
Which version of app you have installed?

lullabyforkful, 2022-02-28:
> Please attach video. 
> Which model of BMS are using?
> Which version of app you have installed?
> 
> 

[image]

smagicld, 2022-07-21:
missing details. closing.

closed 2022-07-21 by smagicld

lullabyforkful, 2022-07-21:
What details?

smagicld, 2022-07-21:
Which model of BMS are using? 
Please attach video or step-by-step guide so we can try to reproduce your issue.