smagicld/xiaoxiangBMS #176: Can't read configuration with custom hardware pin

NeariX67, 2022-03-06:
Changed pin from 000000 to 444444 for example. Reading the configuration will not work.

smagicld, 2022-03-06:
Maybe more details?
Which app version are you using?

You can provide also some logs:
1. go to info screen.
2. tap on build number once
3. go and connect to bms
4. go back to info screen and tap again build number.
5. send me the logs.

NeariX67, 2022-03-06:
Initially worked with the default pin. No progress after pin change
[debug-1.zip](https://github.com/smagicld/xiaoxiangBMS/files/8193352/debug-1.zip)

smagicld, 2022-03-06:
I see you are changing the PIN from 111111 to 444444 with success.
Tried to reproduce the same on my test BMS's (four). can not reproduce your issue.

which version of the app are you using?

NeariX67, 2022-03-06:
1.2.32 6921, should be latest from app store

NeariX67, 2022-03-06:
I accidently mixed up #177 and #176 so i tried to read the configuration instead of changing mos status, so don't be confused about the logs. But it still proves that the hardware pin variable has the old pin in storage until you go back to the device list and reconnect to the bms.

closed 2022-07-21 by smagicld