smagicld/xiaoxiangBMS #177: Unable to change MOS Status after settings hardware pin (even with default pin)

NeariX67, 2022-03-06:


smagicld, 2022-03-06:
This is correct when your BMS is protected with hardware PIN.

closed 2022-03-06 by smagicld

NeariX67, 2022-03-06:
I can't seem to get the reason why that should be the case but okay.
It is inconsistent. Sometimes i can change MOS Status after hardware pin change, some
times i can't.

[debug.zip](https://github.com/smagicld/xiaoxiangBMS/files/8193345/debug.zip)

smagicld, 2022-03-06:
your pin (form the log) is 000000 and the BMS response is OK to it.
The response on MOSFET status change is also confirmed from BMS.

Your problem must be somewhere else.

NeariX67, 2022-03-06:
Both #176 and this issue have in common that the app tries to authorize with an older hardware pin than it has been previously set to. I think this is a bug with saving the hardware pin to a local variable.

smagicld, 2022-03-06:
Try to delete the app and install the latest version again.

NeariX67, 2022-03-06:
Same issue.

smagicld, 2022-03-06:
I reviewed the code / re-tested several times and really can not reproduce the problem.

Have you completely deleted and re-installed the app? (because I see in the logs, after you have changed the PIN to 444444 the app tries with 111111 - as you suggested could be something with storing the pin locally on iOS level)

please also try following:
1. go to info screen.
2. tap on build number once
3. go and connect to bms
4. Go to config section
5. set HW Pin to 444444 
6. go back to bms screen
7. Go back again to main screen (where bms’s are listed)
8. Connect to bms screen
9. Try to enable / disable MOSFET
10. Go back to info screen and tap again build number.
11. send me the logs.

based on the logs I'll try to finde the problem.

NeariX67, 2022-03-06:
Sent you an email. This time it worked flawlessly. I'm currently trying again without going back to the device list, but i don't get a hardware pin change confirmation no matter how often i try (it sends the command, no idea what the response is)

NeariX67, 2022-03-06:
Btw, an alert for a wrong hardware pin would be nice. Clicking "OK" with no feedback seems odd to me.

smagicld, 2022-03-06:
> Btw, an alert for a wrong hardware pin would be nice. Clicking "OK" with no feedback seems odd to me.

you mean if you enter wrong HW PIN, press OK button and then the PIN Dialog pop-up again?

smagicld, 2022-03-06:
> Sent you an email. This time it worked flawlessly. I'm currently trying again without going back to the device list, but i don't get a hardware pin change confirmation no matter how often i try (it sends the command, no idea what the response is)

no e-mail received yet....
maybe I can finde something within your logs once I get them.

NeariX67, 2022-03-06:
Gmail seems to report those zip files as a virus. Have 3 mails in my outbox and no indicated progress since hours.

smagicld, 2022-03-06:
can you upload it here?

NeariX67, 2022-03-06:
[debug.zip](https://github.com/smagicld/xiaoxiangBMS/files/8193510/debug.zip)
[debug.zip](https://github.com/smagicld/xiaoxiangBMS/files/8193511/debug.zip)