smagicld/xiaoxiangBMS #178: Balancing whilst discharging

Jaco-TTT, 2022-03-27:
Sometimes when I write values to the BMS, although the option to "Balance whilst charging" is set, it also balances whilst discharging.

Using this app and the Andriod App, cannot seem to get it to stop balancing whilst charging.

Sometimes it works if I disconnect and re-connect the BMS.
Mostly I have to keep on trying to set the Balancing settings on/off a few times.

It "feels" like the BMS is ignoring the setting.

smagicld, 2022-07-21:
the App does not control balancing - it is the FW of BMS. please ask your reseller for support on that issue.

closed 2022-07-21 by smagicld

Jaco-TTT, 2022-07-22:
> the App does not control balancing - it is the FW of BMS. please ask your reseller for support on that issue.

Correct, BMS does the balancing, the App just enables/disables the setting.

BMS supplier says it is working, it is the App. 
App supplier says it is working, it is the BMS firmware. 

I can see the BMS side works when I increase the balancing start volts to like 3.45v, it then does not balance discharging below 3.45v.

How do I know the App actually sets the setting on the BMS firmware? 

The BMS I'm currently having the issue on, is the JBD AP20S004-P20S-200A-200A-B-U
Works on JBD-SP04S020 and intermittently on the JBD-AP20S004 model.

smagicld, 2022-07-22:
I repeat: the app does nothing with balancing - it just reads/writes settings.
how to check? very simple - read out settings again and you will see the value stored in BMS.