smagicld/xiaoxiangBMS #179: Settings for over 100v batteries

Pristik, 2022-03-29:
Hello, well done app!
The only one problem I met: 
I have 30s lifepo4 battery and I need to change the values in “Batt over/under voltage” sections, but… there are only 5 digits can be stored. As you understand in 30s battery this values have to be 6 digits in mV.
Can you help to solve this problem?

smagicld, 2022-03-29:
Hi

Send me your email - I’ll add you into the tester group. 
I‘ll extend it to 6 digits and let you know once the Test build will be approved by apple.

Pristik, 2022-03-30:
Hi, thanks a lot!
Here is it spristavko@me.com

smagicld, 2022-03-30:
Ok. Once apple releases the build for testing you’ll get TestFLight invitation. 
Your build is: 1.2.33 (6953)

smagicld, 2022-04-11:
delivered with 1.2.34

closed 2022-04-11 by smagicld