smagicld/xiaoxiangBMS #180: BMS Werkseinstellungen

Ni2032, 2022-04-02:
Hallo,
Wie kann man das BMS zurücksetzen. Weiß jemand wieviel Volt der Akku LiFe4 Bei 80%,60%, 40% und 20% hat. Das der SOC genau stimmt. Vielen Dank.

Ni2032, 2022-04-02:
13,6v 4 Zellen sind es.

smagicld, 2022-04-02:
Hi
Es gibt keine Werkseinstellungen. Diese Werte musst du anhand der Datasheets deiner Zellen setzen. Dort findest du die Entladekurven (Charts Zellenspannung vs. SOC). Diese Infos kannst du sicher von deinem BMS / Zellen Verkäufer verlangen.

closed 2022-04-02 by smagicld

Ni2032, 2022-04-02:
Habe ich nicht da es ein Akku Pak ist. 105Ah ich habe da nun ein ordentliches BMS verbaut. Die Daten brauche ich gibt es da keine Standard Volt werte wo sich der Akku bei wieviel Volt befindet…

smagicld, 2022-04-02:
Am besten fragst du beim Verkäufer/ Hersteller - sie wissen es genau.

Ni2032, 2022-04-02:
Geht nicht. Und wen es möglich wäre wen einer schreibt der es weiß und für einen 105Ah Akku schon eingestellt hat.

smagicld, 2022-04-02:
Sorry, aber du verstehst es nicht: jeder Hersteller / jede Zelle ist anders! Viele Zellen werden auch nach Einsatzzweck gebaut (z.B. temp -20 grad). Schau dir Datenblätter von verschiedenen Hersteller oder Modellen. Niemand wird dir irgendwelche Werte geben die 100% korrekt sind. Du kannst dir aber Irgendwelchen Datenblatt nahem und dort die Werte auslesen in der Hoffnung es passt für deine Zellen.

Ni2032, 2022-04-02:
[image]

Ni2032, 2022-04-02:
[image]

Ni2032, 2022-04-02:
Wie gesagt war das ein Akku Koffer den ich als gebraucht gekauft habe. Daher habe ich keinen Hersteller. Der QR Code geht nicht mehr. Aber wer den selben Akku hat und die werte hat kann die gern einstellen. Der Code ist ja bei allen identisch.

Ni2032, 2022-04-03:
Das BMs Balanciert sehr schlecht unterschied 0,5volt!

smagicld, 2022-04-03:
Dann musst du beim Verkäufer / Hersteller des BMS boards reklamieren. Wir können dir nicht helfen. Die App zeigt an was das HW meldet.

Ni2032, 2022-04-03:
Reklamieren bei Ali:-))))) niemals! Kenne keinen der recht bekommt. Entweder es geht oder Pech!