smagicld/xiaoxiangBMS #181: SOC by voltage?

Kwintyy, 2022-04-06:
Would it be possible to show the SOC by the % voltages given in the settings list? Instead of relying on the coulomb counter in the BMS? I often use the BMS just for charging only and this is therefore  incorrectly showing its state with the coulomb counter if i discharged outside of the BMS.

Thans.

smagicld, 2022-04-06:
It would be possible, but from my experience with batteries: It is not possible to calculate SOC based on voltage except you have constant drawing. every current increase / decrease affects the voltage.
I'm afraid this feature would bring lot of complains.

Kwintyy, 2022-04-06:
I agree with you that voltage indeed is not a great way to monitor its percentage. But if you would only use the voltage table filled in at the settings page when the voltage fluctuation is minimal. I think the results should be good enough. I just have a lot of cases where the load is outside of the bms due to its high current. And now i end up with an empty battery in voltage still showing a 100% icon. 

A extra feature for the V2 app maybe? Adding the lowest/highest cell voltages in the BMS overview list?

smagicld, 2022-04-07:
Hi 

I see your case. But why don't you go over BMS discharging MOS? you can have them up to 200A so this should be not an issue (I have my solar battery 48V where I discharge on Peak up to 100A: https://solar.deszynski.com). If you give me good reason why you do not discharge over discharge port, then I could implement it in the v2 of the App.

Kwintyy, 2022-04-07:
Hi,

Nice solar page you created!  
I'll sum up a few points. 

Safety:
At first I did this to always to have backup, I use the battery in a foil and don't want my BMS to turn off my output. I rather drain the battery further or even kill i and save my ass from a crazy boat, windstorm or whatever. How do I protect the battery now you might ask. I measure the separate BMS output with an Arduino and cut my PWM signal in half if the voltage on this pin disappears. 

Cost,
My currents are in the 100 to 150A continues and getting a decent BMS for this is much more cost full than the 20A version I use now for charging only. 

Reliability:
Since im using all these things in and around the water its not impossible to have a BMS measure a strange cel voltage deu to terminal corrosion or a wet circuitboard. Non the less of the DIY potted BMS and a waterproof battery and a waterproof compartment I can get wet if you where to have a bad seal and a bad potting. With the idea of "less is more" I took the BMS out of the discharge circuit to remove a possible failing component.

Heat:
Also getting the generated MOSFET heat out of the waterproof casing and potted  BMS is a hassle.

And an image for the sake of sharing. Getting a malfunction 10 to 15k from your starting point is crap even if you are foiling together. :P
[image]

smagicld, 2022-04-11:
Hi

OK, I see your points. 
Just for fun I'll implement it within the BMS Tool.

Short spec:
1. Configurable for each BMS separately (BMS -> Config -> Voltage SOC)
2. It will support 5 Levels Voltage: For 20, 40, 60 80 and 100%

Any other features?

Kwintyy, 2022-04-12:
Hi,

That would be great! 
I would love to see the highest and lowest voltages on the BMS overview page.

smagicld, 2022-05-16:
Highest / lowest voltage is since beginning on the BMS overview page:
[image]

Kwintyy, 2022-05-16:
> By any chance already implemented ?
With this I meant the SOC by voltage.

smagicld, 2022-05-17:
> 

It’s still under development, will come with next release.

Kwintyy, 2022-05-17:
Fantastic!