smagicld/xiaoxiangBMS #183: Bely BMS Not reachable

Timmy69Ms, 2022-05-07:
Hi 
I have 2 new Bely 150ah here. I can see them and select, But on 2nd Screen there is no Date
[image]
[image]. Stays in Connection mode
[image]

Timmy69Ms, 2022-05-09:
After deletion of App and new Installation ive had one Time a Connection…, But no fully success.

smagicld, 2022-07-21:
Belly BMS is not supported.

closed 2022-07-21 by smagicld