smagicld/xiaoxiangBMS #184: NO current calibration

Kwintyy, 2022-05-13:
I have noticed that calibrating the charge current takes a long time and is in consistent can take up to 5min.
The discharge current is not responding even after 30min of waiting till the setting is applied.

Currents are stable. And I’m using a JBD-SP15S030 the well known Chinese type.

smagicld, 2022-05-13:
What is your calibration procedure?
It takes only some seconds. Measure the current with calibrated multimeter (like Fluke), enter measured value into the field and press the calibrate button. That’s it.

Kwintyy, 2022-05-16:
Enter the new value and then wait till its applied, there is no calibrate or save button.

smagicld, 2022-05-16:
Just enter the measured value (from multimeter display) and press the marked button.

[image]

closed 2022-07-21 by smagicld