smagicld/xiaoxiangBMS #185: Simultaneous In/Out amps

AW123AW, 2022-05-15:
Hello
When charging and discharging at the same time, the charge and discharge amp readings disappear.  For example, charging the battery and pulling a load. Readings work only if doing one at a time.  

Thank you, 
Adam

TheDodgersOnline, 2022-05-15:
HiWith mine the charge current goes to power the load first and what is not required then goes to charge the battery - so if the load and charge current are the same there will be no readings. If the charge is greater than the load there will be charge readings.Regards

smagicld, 2022-05-15:
As theDodgers commented: if you charge and discharge with the same current, the BMS board will report 0. 
this is basic physics- first Kirchhoff law: https://en.m.wikipedia.org/wiki/Kirchhoff%27s_circuit_laws

closed 2022-05-15 by smagicld